BitBoxSwiss/bitbox02-firmware #1155: Coin Integration: Solana (SOL)

belg1o, 2023-11-20:
# Coin Integration Request: Solana

## Description
Solana a rising blockchain in the L1 space has been around for more than 3 years, nearing the 4th year. It has survived a full cycle of interest, and is gaining market traction again. I'd like BitBox to consider adding support for this token as we see more daily active addresses active on the network. 

## Coin Information
- **Coin Name:** Solana
- **Ticker Symbol:** SOL
- **GitHub Repository (if available):** https://github.com/solana-labs

## Additional Information
Sustained active network over the last two year period. 
[image]

## Supporting Documentation
/

benma, 2023-11-21:
Hi

We are currently not planning on adding support for SOL - we don't see consistent and large demand for it at the moment. If we notice increased demand for a particular coin we will consider adding it.

belg1o, 2023-11-21:
> Hi
> 
> We are currently not planning on adding support for SOL - we don't see consistent and large demand for it at the moment. If we notice increased demand for a particular coin we will consider adding it.

BitBox is a pretty niche wallet; up until recently it was impossible to use in DeFi, it is only thanks to Rabby we are seeing traction there. Taking that leads to the following:

It's unlikely vocality will increase until more DeFi users come in, people are more likely to purchase hardware wallets that already support these ecosystems. The reason other hardware wallet providers [incl. niche ones such as GridPlus] support these is because there is clearly demand for them. 

Will leave it up to you when to implement it, but its hard to measure demand for an item when your userbase is already a small cohort within another cohort.

twin-wolf, 2024-02-11:
Add me to the list of people that would really like to see SOL added to Bitbox02.   I am bringing over 8 people, each with an old Ledger and all have SOL in their accounts.   The only downside I have seen so far to Bitbox02 is the lack of SOL support.   SOL is #5 by market cap, beating out LINK and Litecoin.   

I hope you will reconsider and add SOL.   I understand the demands and additional resources to support more coins but certainly this one is both long term and worthy of support by a company like Bitbox02.

Yzord, 2024-03-08:
> Hi
> 
> We are currently not planning on adding support for SOL - we don't see consistent and large demand for it at the moment. If we notice increased demand for a particular coin we will consider adding it.

You should not even consider to add support for what coin. You try to be open, but as a dev you are already blocking certain coins. SOL is the number 5 coin and there should not even be a debate to add it or not. The way you look at certain things is one which decides me not to buy this wallet. It's not about demand or consistency, it is about the trust you want to spread as a team behind this wallet. And the forced view on certain aspects gives me a red flag.

LevSky22, 2024-05-17:
+1 for support

schmitmd, 2024-07-14:
If we're arguing that SOL deserves addition simply based on market cap, I would point out that Terra/LUNA once held 5th place too..

If we're arguing technical stability/readiness, that's different, but adding another L1 blockchain to the firmware is a massive amount of work, not to mention introducing some sizable technical responsibility for maintaining/updating it going forward. 

As I understand it, the dev team for BitBox is quite small and they've been able to keep things tight and functional by choosing security and quality over an extensive feature set.  If you disagree with those priorities, then maybe this is indeed not the wallet for you (yet).  

I think that the best way to nudge the devs into making this work is to put your money where your mouth is by making a bounty.  That shows the interest benma spoke of in the first comment a lot more concretely than "me too!" posts in GitHub.  Maybe it'll help BitBoxSwiss contract/hire the talent needed to get it done.

belg1o, 2024-07-14:
> If we're arguing that SOL deserves addition simply based on market cap, I would point out that Terra/LUNA once held 5th place too..
>
> If we're arguing technical stability/readiness, that's different, [...]

SOL ranks highly in terms of market cap, people using the network and devs exploring opportunities. 

Anyway, I've stopped caring about BitBox. I've returned to using Ledgers and now BitBox turned into a backup wallet. Will probably get a Grid once they roll out their portable wallet next yearish. 

Development of features on BitBox is just way to slow compared to other players in the market.

schmitmd, 2024-07-14:
I wasn't arguing _against_ SOL.  I think it's got potential; I just wanted to maybe clarify why it's not already a thing and why it's not been added to the BitBox roadmap.  If you know a Rust dev who's willing to get their hands into the codebase, a Pull Request might be another way to get it added.

poelzi, 2024-07-16:
solana is the future of blockchains and not supporting this is a full blocker for me.
solana is not just another L1 block chain, it's design is just brilliant and will replace many L2 chains simply by being fast and cheap enough. There are dapps spawning up that solve many common problems like distributed storage.

andrasmining, 2024-08-08:
I would like to see SOL in the wallet as well, but I am also **rather safe then sorry**. Just think about it a second what it means if you add another blockchain to the firmware (I know its hard, but try it). Therefore @Devs, take your time, think it through and only add it if it really is safe and ready.

FeowM, 2025-01-28:
Hello BitBox team, @benma,
please add the support for SOL. Otherwise I have to leave my BitBox... That would be a pity...
Thanks in advance. :)