BitBoxSwiss/bitbox02-firmware #1157: Custom Firmware Attestation Question

dylangerdaly, 2023-12-10:
Hello,

I'd like to buy a Bitbox02 however I'd like to know how compiling (and signing?) the firmware myself affects Device Attestation, also I can't seem to find any documentation on signing releases

Thank you

P.S - Pls support Monero, that'd be a massive gain imo.

benma, 2023-12-10:
Hi

Compiling the firmware does not affect the device attestation.

Compiling the firmware let's you verify that the official release was created from the source code in this repo. See https://github.com/digitalbitbox/bitbox02-firmware/tree/master/releases#reproducible-builds

dylangerdaly, 2023-12-10:
Thank you for that.

If I were to tap the I2C bus between the Secure Element and MCU, while the seed is being unlocked, would that leak the seed? Is there any form of transport security between the SE and MCU?

benma, 2023-12-11:
You can't boot modified firmware - the BitBox02 only boots firmware signed by engineers at Shift Crypto. What is it that you plan on working on? 

If you tap the I2C bus, the seed would not be leaked. The transport is encrypted for many chip commands, but the primary reason it won't leak the seed is that the seed is never even transmitted there. See this article for some more details: https://bitbox.swiss/blog/best-of-both-worlds-using-a-secure-chip-with-open-source-firmware/

dylangerdaly, 2023-12-12:
Ah - yeah this is what I'd like to work on, can I buy a non-blown Bitbox02?

I'm guessing the MCU has some form of Secure Boot on it, I'd prefer to do that than to re-work a black MCU onto the device.

Ah thank you, very thorough docs!

dylangerdaly, 2023-12-12:
I'm willing to pay a little more for an unblown device, I just want to control the firmware myself.

benma, 2023-12-13:
@dylangerdaly can you elaborate on what customizations you want to work on? Or is your goal simply to be able to build and install the firmware from source as-is without modifications?

dylangerdaly, 2023-12-13:
My goal is to control the RoT of my own hardware (wallet)

Outside of that, implement OpenPGP support among other things that I shouldn't really need to justify.

The ability to boot my own firmware would be a major differentiator between Bitbox02 and a Ledger.

Is the hardware open? Am I able to order a Bitbox02 without the `BOOTLOADER_PRODUCTION` ifdef set? If this isn't possible, can I order a Bitbox02 **_without_** the MCU soldered to the board? That way I can solder a fresh untouched MCU.

Thank you

lachgil, 2023-12-15:
+1 to rolling our own firmware, especially with all the recent issues with ledger it would be great to finally have full control over our own hardware.

dylangerdaly, 2023-12-17:
Any update on this?

benma, 2023-12-18:
Yes - unfortunately it is not possible. We don't produce and sell unlocked devices, and we don't take custom orders for this either. Sorry to disappoint.

dylangerdaly, 2023-12-18:
No worries, am I able to take the device apart and re-work (re-solder) the ATSAMD51J20A MCU? This can be difficult when the board is epoxied.

Are the hardware docs enough for me to send them thru to a fab in China, I'd like to produce a Bitbox02 that hasn't been blown, the schematics and BOM cost docs should be enough to produce this?

How does Bitbox personalize a blank device? I'd guess thru some sort of pogo pins on the board.

benma, 2023-12-19:
A blank device is setup using the factory-setup firmware (part of this repo).

> Are the hardware docs enough for me to send them thru to a fab in China, I'd like to produce a Bitbox02 that hasn't been blown, the schematics and BOM cost docs should be enough to produce this?

I am not familiar with the hardware side. Maybe @jadzeidan can give some input.

dylangerdaly, 2023-12-19:
Ahhh thank you, so I see an attestation keypair is generated on-device (Within the SE) and returned to you.

https://github.com/digitalbitbox/bitbox02-firmware/blob/master/src/factorysetup.c#L152-L168

Using EMFI or Vcc Glitching I should be able to flip the MCU into a debuggable state giving me the attestation keypair, are you sure I can't just buy an unprovisioned device? Valid Attestation keys would let me run my own firmware _and_ say everything is fine to the client app, that seems like something Bitbox wouldn't want

benma, 2023-12-20:
The keypair generated there is not a valid attestation, for it to become valid a certificate needs to be set using the OP_SET_CERTIFICATE command. Official BitBox02s already have such a certificate anyway, so you would not even need to make a new keypair.

Re unprovisioned device: can't do it, but it would definitely be interesting to offer this in the future. Maybe next year :crossed_fingers: 

Please contact security@bitbox.swiss regarding your glitching attempts, we'd be interested in learning more about it.

dylangerdaly, 2023-12-21:
Ah I see, well that would work perfectly for me, passing attestation with my own firmware is even better!

Is it possible the Bitbox03 would see Unlocked SKUs? Feel like I should shout out the Precursor https://www.crowdsupply.com/sutajio-kosagi/precursor they're working on turning the FPGA into a RISC-V SoC.

Will do, I'll email and update here on progress.

closed 2023-12-21 by dylangerdaly