BitBoxSwiss/bitbox02-firmware #1203: Electrum can't connect to bitbox

v-barros, 2024-04-24:
When trying to connect the electrum wallet to my bitbox app, I see this error message:

```
Traceback (most recent call last):
  File "/home/vini/.local/lib/python3.10/site-packages/electrum/plugins/bitbox02/qt.py", line 143, in check_task
    self.client.pairing_dialog()
  File "/home/vini/.local/lib/python3.10/site-packages/electrum/plugin.py", line 406, in wrapper
    return run_in_hwd_thread(partial(func, *args, **kwargs))
  File "/home/vini/.local/lib/python3.10/site-packages/electrum/plugin.py", line 399, in run_in_hwd_thread
    return fut.result()
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 458, in result
    return self.__get_result()
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 403, in __get_result
    raise self._exception
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/vini/.local/lib/python3.10/site-packages/electrum/plugins/bitbox02/bitbox02.py", line 177, in pairing_dialog
    hid_device.open_path(self.bitbox_hid_info["path"])
  File "hid.pyx", line 157, in hid.device.open_path
OSError: open failed

```
This error appears after clicking on 'next'
[image]

[image]

I've made the udev settings before 
[image]

And the libraries are installed too
[image]
I'm using
BitboxApp version 4.39.0
Electrum version 4.5.4
Ubuntu 22.04.4 LTS

benma, 2024-04-25:
Try closing the BitBoxApp and any other wallets/software that connect to the BitBox before using Electrum. The BitBox can only handle one wallet connection at a time.

benma, 2024-07-12:
Closing due to inactivity. Please comment or re-open if the problem persists.

closed 2024-07-12 by benma