BitBoxSwiss/bitbox02-firmware #1213: Regtest addresses displayed incorrectly

abradshaw, 2024-05-10:
When using the BitBox02 wit the regtest network, the destination address displayed on the device is incorrect. It shows a testnet address instead (start and end parts are different - middle part is the same)

benma, 2024-05-12:
What software are you using?

The BitBox02 does not currently have support for regtest, but we can definitely add support for it if there is demand.

I assume you are are talking about bech32 addresses?

abradshaw, 2024-05-15:
I am using Liana wallet with Regtest. 

The address I send to starts like this `bcrt1qjj0kef7p8hwp28xcjsg3azyzzg6369j...`

But the address shown on the device is a testnet address `tb1qjj........`

If I ignore this and sign it anyway, the transaction goes through to the right address 

I have attached and example pbst that you can load into bip174.org

[psbt.txt](https://github.com/digitalbitbox/bitbox02-firmware/files/15319537/psbt.txt)

If you chose regtest in the pulldown on the top right you see the correct address - if you chose testnet in the pulldown you see the address displayed on the bitbox

benma, 2024-05-15:
Seems like Liana simply chooses testnet for the BB02 even if it's regtest, because the BB02 does not support regtest.

So how important is this feature to you? I think you might be the first person to use regtest with the BitBox02 for actual testing :smile:

abradshaw, 2024-05-15:
Well I eman if it is just for one person I can see that this is not a good use of time :-)

I can live with just ignoring the address - well checkin the middle part of the address which seems to stay the same

I shall close this - thanks for the info though

closed 2024-05-15 by abradshaw