BitBoxSwiss/bitbox02-firmware #1251: [Bounty] Implement sending to Silent Payment addresses - 0.01 BTC

sethforprivacy, 2024-07-12:
A bounty has been opened on silentpayments.xyz from a trusted, anonymous source to implement sending support in BitBox firmware to Silent Payment addresses.

I wanted to open an issue here to raise awareness and ensure that prospective contributors are aware of open bounties for BitBox from outside sources.

You can learn more about the bounty at https://silentpayments.xyz/docs/bounties/ and find more development resources focused on Silent Payments at https://silentpayments.xyz/docs/developers/.

josibake, 2024-07-17:
> [DLEQ proofs](https://x.com/josibake/status/1800932941750522210) would solve this, but I could not find a suitable implementation of DLEQ for secp256k1, so that solution could be quite a while out (cc @josibake, please correct me if I am wrong).

I also haven't found a good implementation for secp256k1, but if it helps there is a draft proposal here which fully specifies the algorithm: https://gist.github.com/andrewtoth/df97c3260cc8d12f09d3855ee61322ea

benma, 2024-07-17:
@josibake the thread you linked points to an implementation in secp256k1-zpk: :exploding_head: 

https://github.com/BlockstreamResearch/secp256k1-zkp/blob/master/src/modules/ecdsa_adaptor/dleq_impl.h#L1

josibake, 2024-07-17:
ah nice! i hadn’t seen that comment

benma, 2024-09-19:
Support for this is now merged:

https://github.com/BitBoxSwiss/bitbox02-firmware/pull/1220

With support in the BitBoxApp also merged: https://github.com/BitBoxSwiss/bitbox-wallet-app/pull/2922

The output verification on the host using DLEQ proofs mentioned above is implemented was implemented in https://github.com/BitBoxSwiss/bitbox02-api-go/pull/105.

This is projected to be released in BitBoxApp v4.45 and firmware v9.21.

benma, 2024-09-19:
@sethforprivacy how do I claim the bounty? :grinning:

closed 2024-09-19 by benma

josibake, 2024-09-20:
amazing work, @benma !

sethforprivacy, 2024-09-20:
> @sethforprivacy how do I claim the bounty? 😀

Can you share a Signal username, or can I send you a SimpleX address to connect to for coordinating payment? I'll connect you with the bounty provider :)

benma, 2024-09-20:
I decided to not claim the bounty, and asked them to send it to https://brink.dev/donate instead, which they did.