BitBoxSwiss/bitbox02-firmware #1254: ASCII extended character set disappeared after update to last firmware

wilya7, 2024-07-13:
Fyi, the feature is now released in firmware v4.2.1, which can be installed with the [BitBoxApp](https://shiftcrypto.ch/start/).

_Originally posted by @benma in https://github.com/BitBoxSwiss/bitbox02-firmware/issues/151#issuecomment-547441007_

wilya7, 2024-07-13:
This is a serious problem. Since last firmware release, the ASCII extended character set is not anymore available. This makes accessing the wallet impossible for anyone adopting the 25th word with extended characters set.

wilya7, 2024-07-13:
My apologies. I made a stupid mistake. The character set is still there. Closing the issue.

closed 2024-07-13 by wilya7

benma, 2024-07-13:
You may have confused the device password with the optional passphrase. Only the optional passphrase has a special chars keyboard.