BitBoxSwiss/bitbox02-firmware #1307: Security concept of the secure element

MaJerle, 2024-09-21:
I've already asked a question before, which was not fully answered, but partially. [Link to first discussion](https://github.com/BitBoxSwiss/bitbox02-firmware/issues/1160)

I'm reading the BitBox02 security concept again. There is a nice picture on your blog:
[image]

In the previous issue (link above), @benma mentioned:
> The answer lies in the picture you added actually. The secure chip receives the hash of your device password (so not your password), and applies KDF to it with internal keys. The output is then again hmac'd with your password, which results in the final encryption key. This means that the secure chip (and the communication line to the MCU) never have the seed nor the encryption key of the encrypted seed.

In your blog, you claim that if an attacker has access and performs reverse engineering in a country where it is legal, you want to be protected from this. Me, as an attacker, if I sniff I2C lines:

- I get `sha256(S || P)`
- Secure chip will send back `K`, that is derived from the `KDF(sha256(S || P) || secure_chip_random_key)`
- Finally, MCU will perform `hmac(sha256(S || P), K)` with inner and outer keys, that are stored (according to the comment above)

As I understand:
- System is as secure as your password
- Secure element is there only to go through it to prevent fast brute force attacks and to limit to cca 750k attempts.

Finally, in your marketing, you mention there is attestation key for device geniune check.
- Where is the private key stored, that makes sure one cannot clone the device? Is it in the secure element?
- Is secure element preloaded during your manufacturing with device certificate and private key?

benma, 2024-09-22:
> I've already asked a question before, which was not fully answered, but partially. [Link to first discussion](https://github.com/BitBoxSwiss/bitbox02-firmware/issues/1160)
>
> I'm reading the BitBox02 [...]

The device attestation private key is stored in the secure chip. The private key is generated during manufacturing inside the chip. The certificate is created and put on the device flash memory during manufacturing.

benma, 2024-10-14:
I'll close this as it's not actually a firmware issue. Please re-open if I am mistaken. Thanks.

closed 2024-10-14 by benma