BitBoxSwiss/bitbox02-firmware #1318: Bug: Regtest and Signet: Incorrect addresses

andreasgriffin, 2024-10-13:
The regtest descriptor: `wsh(sortedmulti(1,[3bceb0d1/48h/1h/0h/2h]tpubDEfDLjt9KPzMJRAVqD6SDETg6ErTpGcrwiqgoPRiKuVjZn5ZXSnmXdZuj7mQZ8SNqQTZLoiPpcRwCiqQkt7kUUQFNfpifyA9wTnLJ8ChYJT/<0;1>/*,[3dcb0605/48h/1h/0h/2h]tpubDFdsrUN9ZC54Qp3qG7dHK2u16GFjzVtqq4MJY1Srt4vV6BBDmgL5nkBCKguvs4LEt664D1KjMcgaiq66E4iEUwS1hVisKjArGmBtVpxF78C/<0;1>/*))#6dp46rpc`
gives as first address `bcrt1qzsv4g3w6zy5v4usgc9pq2rrfra7nkj0ppsk3w8337ss35tz7rfeq3mpv5m`

When this is validated on the bitbox via hwi  [display_multisig_address](https://hwi.readthedocs.io/en/latest/devices/index.html#hwilib.devices.ledger.LedgerClient.display_multisig_address)  one gets an address that is almost identical, but the last few characters are different:

[image]

This issue is in REGTEST and SIGNET, but not TESTNET3 and mainnet (afaik)

andreasgriffin, 2024-10-13:
I got it... this is actually the testnet3 address.   Bitbox02 doesnt seem to support regtest...

closed 2024-10-13 by andreasgriffin

benma, 2024-10-14:
Thanks. It does since https://github.com/BitBoxSwiss/bitbox02-firmware/releases/tag/firmware%2Fv9.21.0, which will be released with the BitBoxApp soon. 

Which wallet/library are you using? Regtest support might need to be integrated there too.

andreasgriffin, 2024-10-14:
I use bdk (regtest supported) and hwi and wrap it https://github.com/andreasgriffin/bitcoin-usb  for my bitcoin wallet https://github.com/andreasgriffin/bitcoin-safe