BitBoxSwiss/bitbox02-firmware #1324: Repro Builds

j34g, 2024-11-12:
I can see that bitbox has reproducible builds but how can we manually install the specific binaries that we have compiled onto the bitbox itself without having to trust that the app is using the correct firmware.

Basically is there an easy way to interact with the bitbox to manually select the file we want to use to update? If not please make this a feature as it is essential.

j34g, 2024-11-12:
Also please make an update so that the firmware has stays on the screen until you tap the top and bottom so that you can have sufficient time to read and verify it. Thank you :)

NickeZ, 2024-11-12:
We only support signed firmware, so on an official device you will never be able to run your own firmware. But hardware documentation is available and all software is open source, so you can build your own version and run your own firmware.

Using the pinch gesture to confirm the firmwares hash is a feature request we will have to discuss internally before giving any promises.

j34g, 2024-11-12:
> Using the pinch gesture to confirm the firmwares hash is a feature request we will have to discuss internally before giving any promises.

Understandable, but it does seem confusing as to why the feature is there if it is _barely_ usable given the short time it pops up.

NickeZ, 2024-11-12:
> I understand it only accepts signed firmware. Im saying if I build the firmware myself and compare its hash to your signed firmware on the repo I want to be able to move that **specific** file to the bitbox for installation as I have just verified it matches the one I have just built.
> 
> If I cant do this then my verification means nothing as I don't know where the app is getting its firmware from.
> 
> I hope that makes sense.

I don't have this in front of me, but you can binary diff our signed binary with your compiled file and see that they are the same except for the signatures. So flashing your own firmware would be the same as flashing the firmware we provide.

j34g, 2024-11-12:
> I don't have this in front of me, but you can binary diff our signed binary with your compiled file and see that they are the same except for the signatures. So flashing your own firmware would be [...]

I have used the repro builds and verified everything already. That step is done. 

My question is with bitbox we dont know what firmware is being sent to the bitbox specifically due to the streamlining of the bitbox app. I want to know whether there is a way that I can send the **specific** firmware that I have downloaded from the repo and compared with my build to the bitbox.

Just like you can build and verify the hash of a foundation passport firmware and send the SPECIFIC firmware you have verified to the passport (via SD card) to make sure the verified firmware is being installed.

I hope that has further clarified what I am trying to ask.

benma, 2024-11-12:
- You can manually flash the signed firmware using [load_firmware.py](https://github.com/BitBoxSwiss/bitbox02-firmware/blob/master/py/load_firmware.py)
- You can verify that the official signed release in this repo corresponds to your reproduced build using https://github.com/BitBoxSwiss/bitbox02-firmware/blob/master/releases/describe_signed_firmware.py
- You can display the hash of the firmware before boot and compare it to the hash as output by the above tool

Hope that helps

j34g, 2024-11-12:
Is there a way I can send firmware using the app or does it have to be using python scripts.

j34g, 2024-11-12:
> * You can display the hash of the firmware before boot and compare it to the hash as output by the above tool

The issue is that it displays for such a short time it is hard to verify, I suggested that the hash should persist on the screen until you pinch to proceed with login

benma, 2024-11-12:
> Is there a way I can send firmware using the app or does it have to be using python scripts.

The app does not support that, no.

By the way, sending your own firmware build does not help you much with verification anyway. Verifying the hash before boot gives you similar assurances.

> The issue is that it displays for such a short time it is hard to verify, I suggested that the hash should persist on the screen until you pinch to proceed with login

Thanks for the suggestion, it's an interesting idea. It might be difficult, as this is a bootloader feature, and the bootloader currently does not contain any touch related functionality.

If you plan to verify the hash only once after a firmware upgrade and not every time, as a workaround for the short time you could:

- snap a quick picture with your phone and look at that
-  reconnect to get more time

j34g, 2024-11-12:
> The app does not support that, no.

I think that this would be a nice addition, even if it had to be hidden in the settings as an ''advanced'' feature.

> Thanks for the suggestion, it's an interesting idea. It might be difficult, as this is a bootloader feature, and the bootloader currently does not contain any touch related functionality.

I understand, maybe instead it could just be on the screen for maybe 10 ish seconds and then it goes to login automatically (with a little progress bar to not scare people that its taking alot of time). Just a thought.

NickeZ, 2024-11-13:
> ...

Did you not get answers to all of your questions?

closed 2024-11-16 by copy2018