BitBoxSwiss/bitbox02-firmware #1331: Reproduce Firmware

copy2018, 2024-11-28:
Im trying to reproduce the latest btc only firmware using the instructions here: https://bitbox.swiss/blog/how-to-independently-verify-the-bitbox02-firmware/

The build process is sucessfull and i get the hash 36895857c346c1fbd0d206853b7031e985c1a959e9f7f9396a7ae94dfffa9679 which matches the unsigned one on github.

But when I do a hash of the signed btc only firmware on github i get a hash which is different to the '4b851096907dbd168f1d3873f7be1adc8c43d779d3015b3f3bbf5038842980b1' which displays on the bitbox.

To my understanding the hash of the signed firmware should be the same as the hash that displays on the bitbox. Am I wrong here?

Beerosagos, 2024-11-28:
Hi, the hash displayed on the BitBox screen is slightly different. You can verify it following these instructions: https://github.com/BitBoxSwiss/bitbox02-firmware/tree/master/releases?ref=bitbox.swiss#verify-the-hash-as-shown-by-the-bitbox02-at-startup

copy2018, 2024-11-28:
Thank you this worked!