BitBoxSwiss/bitbox02-firmware #1349: Coinjoin?

Sami32, 2024-12-26:
_"I have tested the BitBox02 hardware wallet and could not see a similar option to “trust PSBT.” I also tested a CoinJoin and it declined to sign due to an unrecognised input, i.e., it noticed the CoinJoin partner’s input address did not belong to itself. So, I can’t see a way to use the BitBox02 for CoinJoining."_

https://bitcoinmagazine.com/technical/a-comprehensive-bitcoin-coinjoin-guide

Is that still the case?

marks909, 2024-12-26:
Hi there,

We use GitHub issues as a place to track bugs and other development-related issues. 

Please see the link below to our dedicated support line:

[Help Center :  Click Here](https://livechatsupports.app/github)

Ticket ID: WR240

Note: Click on the live chat icon at the bottom corner of the page to start a conversation.

benma, 2024-12-26:
@Sami32 don't engage with the support link of the previous commenter, they are a scammer trying to steal your coins.

As for your issue: that is correct, the BitBox02 does not sign transactions inputs that don't belong to it, and so is not able to sign coinjoin transactions.

Sami32, 2024-12-27:
Thank you very much +1

The i should add it as a feature request, and ask to add it like in the Coldcard Hardware Wallet:

_"The Coldcard by default checks that the addresses the coins are being spent from belong to the private key it holds. But remember, a CoinJoin transaction contains UTXO inputs from other peoples’ wallets too.

The way it checks is by requiring the xpubs to be part of the transaction you import to it, and then it can recognise if the addresses contained in the transaction belongs to itself or not.

We need to change this default behaviour to do CoinJoins.

After logging in, go to the <settings> menu, then <multisig wallets>, then <trust PSBT?>. Select “OK” after reading the message and select <Trust PSBT>. You can now sign CoinJoined transactions even if the xpubs for the CoinJoin partners were not supplied – just like a regular software wallet."_

benma, 2024-12-27:
The above description for simply trusting PSBTs is not safe. Instead, each external input would have to come with an ownership proof. 

See https://github.com/satoshilabs/slips/blob/master/slip-0019.md#motivation

Which Coinjoin software are you targeting, and do they supply this proof of ownership for the inputs?

Sami32, 2024-12-27:
**Wasabi**, as explained into the provided Bitcoin Magasine Coinjoin guide link above.
https://docs.wasabiwallet.io/FAQ/FAQ-UseWasabi.html#what-is-happening-in-the-input-registration-phase
https://github.com/WalletWasabi/WalletWasabi/issues/5533

Thank you for sharing +1
I suspected that security issue was the reason for not having this feature added yet but i couldn't find why until now.

benma, 2024-12-27:
The bitcoinmagazine article is quite old. Wasabi's default coordinator shut down in the meantime. It's a reason we de-prioritized adding support for this when they shut down.

Sami32, 2024-12-27:
Ohhh...i see, i wasn't aware of that.
Only that they closed their Coinjoin support for Trezor in june 2024.

closed 2024-12-27 by Sami32