BitBoxSwiss/bitbox02-firmware #1351: [Feature request] Distributed Backup support?

Senjuu, 2024-12-29:
As already mentioned by me in #113 I perceive a distributed backup as the last dealbreaker for me switching to the BitBox02.

Also (already mentioned in #113) with the introduced feature of BIP-85 (child keys) my percieved importance for such a features has risen significantly as a lost seed can now not only jeopardize ones own wallet but the wallet of everyone trusting the seed's owner with their wallet's backup (e.g. children, parents, friends (as advertised by the BIP and your blog). Of course this could be migitated by storing multiple copies of the seed but this also increases the chances of one single copy of the seed being compromised and therefor also the derived seeds.

As I myself do not prefer any specific standart for this I do not inded to put such in the title of this FR. But as suggested by @jonathancross a more compatible standard to [SLIP-39](https://github.com/satoshilabs/slips/blob/master/slip-0039.md) could be [SSKR](https://github.com/BlockchainCommons/crypto-commons/blob/master/Docs/sskr-overview.md#for-developers) which already provides [libraries](https://developer.blockchaincommons.com/sskr/#sskr-libraries) in C or Rust. But which it would be could be discussed by more involved person here.

For more arguments wheter multisig is an alternative or an anhancement to distributed backup please see the discussion in #113.

As SSKR is a standard which could be setup seperatly from the BitBox I want now to mention that sharding the BIP-39 seed outside the BitBox could be done, but would need a secure and trusted device (while the BitBox should already be one of the most if not the most trusted device owned) and would not let the user take advantage of the possibility that the BitBox could store the shards on multiple SD-cards for ease of recovery.

Lastly I want to ask for some additional quality of life features if this would be included in the BitBox:
1. Continue Recovery in case of powerloss (not a partial typed in shard just from the last complete shard)
2. Continue a begun recovery without a PC (e.g. just power from a powerbank) to not need to bring all shards together (just go from shard to shard and type them in / read in the SD-card)
3. Continue a "completed" Dry-Run Recovery for the remainding shards to check the whole setup at once

benma, 2024-12-29:
Thanks.

Does SSKR have any adoption?

Fyi, the Rust library should not be used in production as per the [README](https://github.com/BlockchainCommons/bc-sskr-rust/blob/e1bd52f1db62ae75852e3f24b3219bd98a585fb5/README.md#status---community-review).

The C lib may be viable as it has received an audit, though I am not a fan of pulling in a C library.

jonathancross, 2025-01-05:
> Does SSKR have any adoption?

*  Ledger devices have an app to generate & check SSKR shares: [app-seed-tool](https://github.com/LedgerHQ/app-seed-tool?tab=readme-ov-file#generate-shamirs-secret-sharing)
* It is also implemented in the DIY [Lethekit](https://github.com/BlockchainCommons/lethekit?tab=readme-ov-file#blockchain-commons-lethekit) HWW.
* Hopefully more to come :-)