BitBoxSwiss/bitbox02-firmware #1372: [Feature Request] Add ability to change what's displayed on boot

pakovm-git, 2025-02-13:
By default the device says "BitBox" once plugged in.

One reason for this feature to be requested is in case the device gets stolen, not to tip off about the nature of the device.

benma, 2025-02-13:
Why would a thief steal a random USB device not knowing what it is?

If they do and realize it's a BitBox, what is the problem? The device is password protected.

Practically, it is difficult to hide what it is: "BitBox" appears not only in the logo at boot, and in the "See the BitBoxApp" string, but also in the USB descriptor, which contains the things like "bitbox.swiss", "BitBox02", etc. 

I'd close this as wontfix. @jadzeidan ?

jadzeidan, 2025-02-13:
I agree it adds minimal benefit, maybe a bit of privacy if someone wants to google "BitBox" once they see it, but I think if the user wants to be private about their bitbox ownership, then they should hide the device.

However customizing what is displayed on the device is nice for a different reason: personalization, but that is a separate issue, so I agree we can put this as wont-fix.

closed 2025-02-13 by benma