BitBoxSwiss/bitbox02-firmware #1434: Variability in signature size causes fee estimation issues in external wallets

dende-hodl, 2025-05-05:
Hi,

I'm using the BitBox02 with Sparrow Wallet, and I've encountered an issue where transactions with multiple inputs sometimes fail to broadcast when using a 1 sat/vB fee rate.

After the feedback from the Sparrow developer, it seems the issue is due to small variations in the size of the signatures returned by BitBox02. Sparrow estimates the transaction size using standard-length dummy signatures, but if the actual signature is slightly larger, the resulting transaction ends up bigger than expected, and the effective fee rate drops below the selected value, causing the node to reject it.

This doesn't happen with other signers like Coldcard, Bitcoin Core, or Electrum, which return consistent-length signatures.

A similar issue has been discussed here (involving KeepKey rather than BitBox02):
👉 https://github.com/sparrowwallet/sparrow/issues/1611
While the device is different, the underlying cause appears to be the same - slight variability in signature size leading to inaccurate fee estimation.

Would it be possible for BitBox02 to return deterministic signatures with a stable size? That would improve compatibility with external wallets and align with common implementations. It may also help reduce any theoretical fingerprinting or key exfil vectors related to signature size variance.

Thanks for your time and your work on BitBox!

benma, 2025-05-06:
Thanks for your report

I think this is referring to low-R grinding to make sure the R-part of the signature is always encoded using 32 bytes, instead of needing 33 bytes in ~half the sigs.

First, note that this is an issue only for ECDA sigs, i.e. for inputs spending Segwit v0 UTXOs. For Taproot inputs, the sig is always 64 bytes, so there should be no issue there. So one solution for this is to use Taproot.

As for low-R grinding in ECDSA sigs: this is difficult for the BitBox to implement, as we employ the [anti-klepto](https://blog.bitbox.swiss/en/anti-klepto-explained-protection-against-leaking-private-keys/) protocol to mitigate exfiltration of secrets through the signature. The final signature R value is derived in part from randomness contributed by the host, and as a result, the BitBox itself can't grind for low-R values. There are adaptations of the protocol that could still make this possible, but it's not a trivial task to modify a crypto protocol. 

For the above reason, it is not likely we will manage to produce constant sized ECDSA signatures.

NickeZ, 2025-05-06:
If it always is just a matter of 1 byte, then the fee estimation could just assume 33 bytes instead of 32 and it would always work?

closed 2025-05-06 by benma

benma, 2025-05-06:
@NickeZ It could, but then you'd be overpaying when using signers that always save that 1 byte. The overpaying would be quite insignificant. It's in the witness, so this byte has a 4x fee discount factor. You'd be overpaying by ~$1 if the going feerate was 50sats/vB (~50 times higher than today) and you had 100 single-sig inputs and the BTC price was ~$100k.

NickeZ, 2025-05-06:
Aha, I thought it was only applicable when the fee rate was around 1sat/vbyte.

craigraw, 2025-05-06:
> Aha, I thought it was only applicable when the fee rate was around 1sat/vbyte.

It only matters when the chosen fee rate is 1 sat/vb, because of mempool policy rules that do not allow < 1 sat/vb transactions to be submitted/relayed.

benma, 2025-05-06:
@craigraw in this case, if you just assume the 1 extra byte in sigs only if the feerate is set to 1sat/vB, the issue would also be fixed with negligible additional fees, right?

benma, 2025-05-06:
> I guess it all depends on the value of a sat!

It also depends on the percentage of the total tx weight. The difference is probably in the order of promilles. Anyway, it for sure wouldn't be a satisfying fix, and I agree with your comment overall.

NickeZ, 2025-05-06:
We can create an issue internally, but as it looks right now it will take a long time before we have the resources to properly implement a fix.